KinDR007/VictronSmartShunt-ESPHOME #14: Smart shunt data import

livestyle4, 2021-11-03:
hello sir,

For a project, I have to find a way to import individual data from the smart shunt. To get the data dump in the first place I have used python code from Github. Using a VE direct cable I was able to get data every second. The problem is that I only need the voltage, current, power and SOC and not all data that the smart shunt sends every second. Do you have an idea how to only get the right data from the data dump? 

All information that I used to get the data from the smart shunt you can find here karioja/vedirect.

your sincerely,
livestyle4

KinDR007, 2021-11-03:
yes filter only what you need. ve.direct always sends all items

livestyle4, 2021-11-03:
do you have an idea what code I could use to filter the right data

KinDR007, 2021-11-03:
or use my esphome component and read only what you need

KinDR007, 2021-11-03:
> do you have an idea what code I could use to filter the right data 

Yep ... delete other code from python script 😉

livestyle4, 2021-11-03:
do you have an example of how to use your code to get only voltage, current, power and SOC

KinDR007, 2021-11-03:
> do you have an example of how to use your code to get only voltage, current, power and SOC 

Yes define only those sensors that you need look to yaml file in source or example in readme.txt

closed 2021-11-03 by KinDR007