KinDR007/VictronSmartShunt-ESPHOME #18: Battery Temperature

icdragon, 2022-02-05:
Hello, 

First of all, this is awesome.  Thanks for putting this together, worked like a charm.

I have a VictronSmartShunt with a temperature sensor connected to the aux, would it be possible to pull temperature?

Cheers, 
Dick

KinDR007, 2022-02-07:
> Hello, 
> 
> 
> 
> First of all, this is awesome.  Thanks for putting this together, worked like a charm.
> 
> 
> 
> I have a VictronSmartShunt with a temperature sensor connected to the aux, would it be possible to pull temperature?
> 
> 
> 
> Cheers, 
> 
> Dick

hello, nothing is impossible😉 I will try to look at it on the weekend, so please remind me on the telegram https://t.me/KinDR007cz

stefanah, 2022-07-20:
As with icdragon I am really liking using this and would really love to see something similar their request.

In particular I use this in a van so would love to use the Aux to monitor my started battery voltage if you ever find time to implement it.

Cheers,
Stefan

blackscreener, 2022-08-14:
> As with icdragon I am really liking using this and would really love to see something similar their request.
> 
> In particular I use this in a van so would love to use the Aux to monitor my started battery voltage if you ever find time to implement it.
> 
> Cheers, Stefan

I will merge all sensors from Python integration:
https://github.com/birdie1/victron/blob/master/lib/victron_serial/victron_smartshunt.py

yoannchaudet, 2022-12-26:
I opened a [small PR](https://github.com/KinDR007/VictronSmartShunt-ESPHOME/pull/27) to add this 🤗.