KinDR007/VictronSmartShunt-ESPHOME #30: Add MQTT and global throttle?

billjuv, 2023-06-21:
I had no problem adding the setup info and connecting to my MQTT broker to my ESP32 but could not find anywhere that ESPHome would allow me to add a global throttle. Rather than add a throttle to each topic, I ended up using your VictronMPPT-ESPHOME/smartshunt-esp8266-example.yaml and modifying it for an ESP32 (it's working great even though I find the MQTT topic structure to be rather long-winded).

I tried this first as I only have the SmartShunt and was using the ESP32 but it didn't suit my needs without being able to throttle down the runaway messages I was getting.

WZBFL, 2023-08-10:
Same "Poblem" here. On MPPT Code it's possible to use "throttle", but same code on shunt gets an error message.

KinDR007, 2023-08-14:
Use MPPT lib to work with SmartShunts and BMV

closed 2023-08-14 by KinDR007

WZBFL, 2023-08-14:
Good idea, I already swapped this lib. It’s a bit confusing to have two different Von meinem Mobiltelefon gesendet