KinDR007/VictronSmartShunt-ESPHOME #33: model SHU065150050 IP65

magnetus26, 2023-08-11:
has anyone tried to use the new smartshunt model
SHU065150050 IP65 . its waterproof and the ve. direct is a cable instead of  a port

thanks
Miguel

KinDR007, 2023-08-14:
Hi , if I look at the datasheet for [VE.DIRECT ](https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.33.pdf)
it looks identical and should work the same as the non-IP65 version.

closed 2023-08-14 by KinDR007

magnetus26, 2023-08-21:
I got this device and it resets to 100% every single day between 8 am and 9 am, also my soc see some really weird spikes of 900%.
i dont think that is the yaml code but definitely theres something wrong. 

I have connected the posisitve of the smart shunt to the positive of the batterty and the shart shunt in series with the load and the  smartshunt TX to esp8266 D7. no GND to the VE.DIRECT port, but that doesnt eplain why the smart shunt restest to 100% by itself. 

[image]

`my yaml code`

```
esphome:
  name: "${lower_devicename}"
  platform: ESP8266
  board: d1_mini
  comment: "Monitor and a Victron Smart Shunt via TTL"

external_components:
  - source: github://KinDR007/VictronSmartShunt-ESPHOME@main
    refresh: 0s

substitutions:
  lower_devicename: "smartshunt"
  devicename: "SmartShunt"
  config_version: "v2021.06.02.001"
  wifi_fast_connect: "true"
  accuracy: "3"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  manual_ip:
    static_ip: 192.168.0.23
    gateway: 192.168.0.1
    subnet: 255.255.255.0
    dns1: 192.168.0.1
    dns2: 8.8.8.8
 
  ap:
    ssid: "${devicename} Hotspot"
    password: !secret wifi_ap_password

captive_portal:

logger:
  baud_rate: 0

api:

ota:

web_server:
  port: 80

time:
  - platform: homeassistant

uart:
  id: uart_0
  tx_pin: D8
  rx_pin: D7
  baud_rate: 19200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256

victron_smart_shunt:
  uart_id: uart_0

sensor:
  - platform: victron_smart_shunt
    battery_voltage:
      name: "Battery Voltage"
      id: bv
      accuracy_decimals: ${accuracy}

    battery_current:
      name: "Battery Current"
      id: bc 
      accuracy_decimals: ${accuracy}

    instantaneous_power:
      name: "instantaneous power"
      id: instantaneous_power
      filters:
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}

    time_to_go:
      name: "time to go"
      id: time_to_go

    consumed_amp_hours:
      name: "consumed amp hours"
      id: consumed_amp_hours
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    min_battery_voltage:
      name: "Min battery voltage"
      id: min_battery_voltage
      accuracy_decimals: ${accuracy}

    max_battery_voltage:
      name: "Max battery voltage"
      id: max_battery_voltage
      accuracy_decimals: ${accuracy}

    amount_of_charged:
      name: "Amount of charged"
      id:  amount_of_charged   
      filters:
        - multiply: 0.000001
      unit_of_measurement: MWh
      device_class: energy
      state_class: total_increasing
      accuracy_decimals: ${accuracy}
      
    bmv_alarm_text:
      name: "BMV alarm"
      id: bmv_alarm
     
    last_full_charge:
      name: "Time since last full charge"
      id: last_full_charge

    deepest_discharge:
      name: "Depth of the deepest discharge"
      id: deepest_discharge
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    last_discharge:
      name: "Depth of the last discharge"
      id: last_discharge
      unit_of_measurement: Ah
      accuracy_decimals: ${accuracy}

    state_of_charge:
      id: state_of_charge
      name: "SoC"

    discharged_energy:
      name: "Amount of discharged energy"
      id: discharged_energy   
      filters:
        - multiply: 0.000001
      unit_of_measurement: MWh
      device_class: energy
      state_class: total_increasing
      accuracy_decimals: ${accuracy}
      
  - platform: template
    name: " ${devicename} discharge "
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: ${accuracy}
    lambda: |-
          float i = id(consumed_amp_hours).state;
          float e = id(bv).state;
          float w = 0.0;
          if (i < 0) {
            w = i * e;
          }
          return {w * -1} ;
        
     
    
  - platform: template
    name: " ${devicename} charge "
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy
    state_class: total_increasing
    accuracy_decimals: ${accuracy}
    lambda: |-
          float i = 792 + id(consumed_amp_hours).state; 
          float e = id(bv).state;
          float w = 0.0;
          if (i > 0) {
            w =  i * e;
          }
          return {w} ;

  - platform: template
    name: " ${devicename} Rem. Capacity "
    unit_of_measurement: Ah
    device_class: energy
    state_class: total_increasing
    lambda: |-
      if (id(state_of_charge).state) {
        return 792 * (id(state_of_charge).state * 0.01f); 
      }
      return NAN;

switch:
  - platform: restart
    icon: mdi:reload-alert
    name: "${devicename} Restart"
```
Best Regards,
M.

JshStadler, 2023-08-22:
You can try this filter for the spikes, though I think your shunt might be faulty if it resets to 100% unless you have set specific settings that makes it reset to 100% SOC:
`    state_of_charge:
      id: "${name}_${smartshunt}_state_of_charge"
      name: "State of Charge"
      accuracy_decimals: 1
      filters:
      - quantile:
          window_size: 7
          send_every: 4
          send_first_at: 3
          quantile: .9`